Jacotheron/FlashForge-CreatorPro2-PS-Profile #7: Ditto mode: Printing before hotends warm

Nidllah, 2022-12-29:
First of all, thank you for this work you have put into this and it lets me and many others use pruca slicer and it has improved the quality of my prints alot!

If you still are working on this project i want to contribute and improve it!!

I have gotten a little problem when using ditto mode.

I was playing around and leveling the bed and printing left, printing right nozzle while one time it started printing while the hotends where at 160C (set to 240C, printing PETG) and the bed at the desired 70. And it can be either nozzle, it wont wait for any of them, it seems like it waits for the bed only? I got one instance where one hotend was at the desired temp but the other where around 170C.

I dont know how it works, but my current workaround is that i preheat everything and then i start printing.
Is it possible to make it wait for the hotends to heat up aswell?

Nidllah, 2022-12-29:
and just now when i tried it again it started printing when the hotends where 60C. (they where at 70 when i started the print) And the bed where about 50C when i started it. Only the bed whent up to 70 but the hotends never seemed to get anything to get to warm up.

Jacotheron, 2022-12-29:
I will try to maintain this project for as long I use the stock mainboard and software. At some point I will exchange the mainboard for a Duet3D mainboard (as I have done with all my other printers), and from then on maintaining will become harder to do.

You are welcome to contribute to the project by either making suggestions or sending a pull-request.

The issue you are experiencing is weird. I typically only use the preheat option (as it takes a while to heat the bed, and then the nozzles, time I use to confirm the slicing, get it on the card etc). Maybe that is why I never experienced it. 

You can perhaps slice an object in the FF slicer (in the modes that give issues), and first test the file on the printer (to confirm that it does not have the issue), before analyzing the file (I make use of the HxD software for Windows, but any Hex Editor should do - note that Notepad and Notepad++ will struggle due to the binary block). The most probable place for the issue, would be inside the starting g-code section of the file (perhaps see if there is a command that I miss).

Nidllah, 2023-01-27:
Okay, i understand, imma try what you explained. I have been using the dual hotends with the FF slicer and it seems to work as intended, but imma try the same file in both slicers and see if i can see something wierd in the gcode. I will get back to you!

If you would like to check it out (imma try, not used to this stuff) imma leave the 2 gcode files in this comment

[files.zip](https://github.com/Jacotheron/FlashForge-CreatorPro2-PS-Profile/files/10515771/files.zip)