Jacotheron/FlashForge-CreatorPro2-PS-Profile #9: Error code: 1

Marcelllllllllll, 2023-04-04:
Hey, am trying to setup the post processing and i am always hitting this error. 
[image]

Jacotheron, 2023-04-04:
Are you sure you have followed the instructions correctly? Specifically the section for the dependencies (https://github.com/Jacotheron/FlashForge-CreatorPro2-PS-Profile#python--pillow-installation).

Marcelllllllllll, 2023-04-05:
Yeah i am pretty sure. As you can see in the picture the pillow and pip command are both already installed. The directory is the same as the directory used in the Prusa slicer
[image]

Jacotheron, 2023-04-05:
In writing the part below, I noticed in your first image, the post processing script does not have a file extension. Is the script perhaps within the "Postprocessing_Flashforge" folder, or is it the file directly in the root? The script should have file extension of ".py"

Ok, in that case, I am going to ask you to try and debug why it does not work for you (these were how I built the script).

1. In the Vendor file (FlashForge.ini), we need to disable the post processor - for now we want the file before it is post-processed, as we will run the post process script manually on this file. To disable the post processor, simply ensure the "post_process" has an empty value (make a backup of it - step 4 of the installation).
2. Now you can slice a model to test with, and place it where it can easily be accessed by command line (maybe duplicate it, so we don't have to re-slice every time). 
3. Next we need to run the post-processor script manually. I see you make use of Command Prompt, that is fine. Command Prompt usually open in your user directory, thus that is an easy place to have the file (maybe call it "test.gx").
4. Based on the images you have supplied, you would then need to run the following command:
`C:\Python3.10\python.exe C:\Postprocessing_Flashforge(^see comment above) test.gx`

If it works, the text.gx should now be processed and ready to be printed on the machine.

Marcelllllllllll, 2023-04-05:
Yes, i forgot to include the ff-creator-post-processor.py at the end of the path in the Vendor file. Thanks for the help. 
<3

Jacotheron, 2023-04-05:
Ok, glad it was solved. Happy 3d printing.

closed 2023-04-05 by Jacotheron