Jacotheron/FlashForge-CreatorPro2-PS-Profile #11: worked in super slicer but not in prusa 2.6.1

ju5th3r3, 2023-09-18:
Flash Forge under other printers didn't show up in Prusa Slicer. I tried super slicer 2.4.58.5 and it is there and worked to select. Any idea on what I'm missing?

Kingston-C, 2023-09-18:
I believe that Prusa changed the way to read vendors in 2.6+. 

A workaround I found is to place "FlashForge.ini" inside of the PrusaSlicer's AppData folder (typically on Windows at "C:/Users/<username>/AppData/Roaming/PrusaSlicer/cache/vendor/"). Then, in this same location, make a copy of any .idx file and rename this file to “FlashForge.idx”

ju5th3r3, 2023-09-18:
> "C:/Users//AppData/Roaming/PrusaSlicer/cache/vendor/"). Then, in this same

Worked thank you so much. I was scratching my head on it, I’m not totally a
Prusa slicer guy yet so was little lost.
Kn

raphael39, 2023-11-14:
Thank you, had the same Issue!

griffmai, 2024-07-12:
I was running into a similar problem with Prusa Slicer 2.8.0.
I would put the FlashForge.ini and .idx files in .../Roaming/PrusaSlicer/cache/ and .../Roaming/PrusaSlicer/cache/vendor, but as soon as I started Prusa Slicer the files would be deleted.

Below is a workaround that finally ended up working for me: 
-Close PrusaSlicer
-Remove all FlashForge files from any folders in .../Roaming/PrusaSlicer 
-Install [PrusaSlicer 2.6.1](https://objects.githubusercontent.com/github-production-release-asset-2e65be/52882701/06bdfc46-aa7a-4a51-82c0-aacd0117af70?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20240712%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240712T201017Z&X-Amz-Expires=300&X-Amz-Signature=52db56be39b958c1ac9e16562edd964dc94b9bf88f1880eb35240cf83324690b&X-Amz-SignedHeaders=host&actor_id=171724198&key_id=0&repo_id=52882701&response-content-disposition=attachment%3B%20filename%3DPrusaSlicer-2.6.1%2Bwin64-202309060711_signed.zip&response-content-type=application%2Foctet-stream) 
-Follow above steps detailed by @Kingston-C 
> ... place "FlashForge.ini" inside of the PrusaSlicer's AppData folder (typically on Windows at "C:/Users//AppData/Roaming/PrusaSlicer/cache/vendor/"). Then, in this same location, make a copy of any .idx file and rename this file to “FlashForge.idx”
-Start Prusa Slicer and add the printer
-Make sure slicing and exporting work
-Upgrade Prusa to newest version

paynterf, 2024-08-12:
This worked for me, sort of... I UNinstalled PS2.8.0, installed 2.6.0 (didn't actually install it - just ran if from the ZIP file download from PS repo).  Ran the configurator and was able to select 'Flashforge'.  However, in the next page, the printer image was blank.  There is a proper FlashForg.png image supplied with the other stuff, but I can't figure out where it goes.  I have tried searching for BIBO*.png or even *.png with no luck.  Any hints?

TIA,

Frank

paynterf, 2024-08-12:
So, I have my CP2 printer configs working now in PS2.8.0, with two significant limitations
 - I get a one-time 'preset bundle version not found' error each time I launch PS
 - The config wizard can no longer be used, as it hangs for 30-60 sec, then produces the same 'preset bundle version not found' error.  Dismissing the error message hangs the config wizard (and PS).

For anyone interested in all the gory details, I documented it all in an update to my '[FlashForge Creator Pro 2 IDEX Profile for Prusa Slicer 2.3](https://www.fpaynter.com/2021/09/flashforge-creator-pro-2-idex-profile-for-prusa-slicer-2-3/)' post.  I would be interested in any thoughts about what I could do to kill off these last two remaining problems.

Frank

twitnic, 2024-08-30:
It should worked with this changes: https://github.com/Jacotheron/FlashForge-CreatorPro2-PS-Profile/pull/20