Jacotheron/FlashForge-CreatorPro2-PS-Profile #12: using left extruder only makes right extruder heat up at layer 1

avxmw, 2023-10-10:
First I'd like to say thanks for making this possible, other than my Linux Flatpak version of PrusaSlicer not playing nice and I need to call the script by hand, it works fine. Thanks.

However, one bug as stated in the title: when printing and settings temps just for the left extruder, the layer change from 0 to 1 sets the temp for both extruders as the T1 is omitted at this point.

Not sure if the way to fix this is in to be done in the post-process script or somewhere else. Not sure if important, but PS version 2.6.1 here.

Jacotheron, 2023-10-13:
This is a very weird issue, and I have not seen it before. Can you perhaps attach the original gcode from PS, as well as the script's output, so that I can take a look and see where the issue comes from.

avxmw, 2023-10-13:
Sure,
output as done by PS: [Buckle-Bottom-ps.gx.txt](https://github.com/Jacotheron/FlashForge-CreatorPro2-PS-Profile/files/12892557/Buckle-Bottom-ps.gx.txt)  
output after conversion with your script: [Buckle-Bottom-conv.gx.txt](https://github.com/Jacotheron/FlashForge-CreatorPro2-PS-Profile/files/12892568/Buckle-Bottom-conv.gx.txt)

When you ctrl+f for "m104", the hit after changing to layer 1 is the problem.

Jacotheron, 2023-10-13:
I see where the issue comes from.

It is mostly due to the temperature change for the "Other layers" in the Filament Settings. I usually just keep it the same as for first layer.

On the readme for this repository (its home page, I havce explained a bunch of Gcode commands). The relevant part is regarding the "M104" command, and specifically that it requires a "T" parameter to set which Tool to apply it to. Since the single extruder versions of the profiles only define a single extruder, PS does not add the tool number.

The way the starting Gcode fixes this, is by manually defining the temperature (if you look at the Printer Settings > Custom G-code). This, along with the option right below "Emit temperature commands automatically" means that PS will not attempt to set the temperature before starting (and thus our fixed version works).

The post processor script simple creates the header for the file, in a way that the printer will understand it and thus allowing us to use the different printing modes. It does not modify the actual gcode of the print, and after the recent optimizations, it will try to completely skip the body of the gcode file (only reading the start/end that contains information that is used for the header).

Something that can be done, is to use "G-code substitutions" (can be found on "Print Settings" > "Output Options"). Using this you can have it search for "M104 S230 ; set temperature" and have it replace with "M104 S230 T1 ; set temperature"
Maybe even using the Regular Expression, catch the temperature (S230 part), and insert it in the replace (so you only have to have one of these substitutions).
Just a note: These changes should be saved per Printer Profile (since the dual and other single head modes will either not require it, or require the other tool). This could create a bunch of extra Print Profiles, which might not be ideal.

Jacotheron, 2023-10-13:
I believe that most people use 1 temperature per nozzle for the whole print, and using more temperatures is basically only where we would experience this.

Also, while having the second hotend heat up, is a mild inconvenience, and would probably not cause any issues.

Therefore, if you want it to behave like you mentioned, I would advise to setup the substitution in PrusaSlicer. If using only the right hotend, you might also experience this issue, since M104 without the Tool parameter sets both hotends. Alternatively to edit the output file (before or after processing) when required.