Jacotheron/FlashForge-CreatorPro2-PS-Profile #17: Maybe an issue with the latest update

Xeet04, 2024-08-07:
Hi,
First of all thank you for making this preset, I've struggled a lot when i tried doing it myself, but an issue I have encountered is that it gives me a pop-up:
"Preset bundle `FlashForge` version not found in index: 1.0.0"
When I'm done with the installation and it's not showing up in the configuration wizard, is this a known issue or have I done something wrong?

fullc360, 2024-08-18:
From what I've gathered about the pop-up you can ignore it. Regarding FlashForge not showing in the wizard I figured out that you need to enter `repo_id = non-prusa-fff` under [vendor] (line 3). Please forgive me if this is crappy formatting, literally created an account to pass on this knowledge!

Xeet04, 2024-08-18:
That seems to have done it, thank you so much, and no worries about any formatting, small issues are better than this not being on the internet.

closed 2024-08-18 by Xeet04